ela-compil/BACnet #2: OBJECT_BINARY_VALUE write data error

chenzx1020, 2017-04-27:
Hello, first of all thank you for your provide BACnet for C#,I met problem in use process, would you please help me to look at. I use OBJECT_BINARY_VALUE  write data happen error: Error from device: ERROR_CLASS_PROPERTY - ERROR_CODE_INVALID_DATA_TYPE, thank you very much!
`            BacnetObjectId oBacnetObjectId = new BacnetObjectId();
            oBacnetObjectId.Type = BacnetObjectTypes.OBJECT_BINARY_VALUE;
            oBacnetObjectId.Instance = 123456;
            bool bIsWrite = oBacnetClient.WritePropertyRequest(oBacnetAddress, oBacnetObjectId, BacnetPropertyIds.PROP_PRESENT_VALUE, new List<BacnetValue> { new BacnetValue(true) });`

gralin, 2017-04-27:
Hi, set the data type for value, right now it's being sent with `BACNET_APPLICATION_TAG_NULL`. Try this:

```csharp
new BacnetValue(BacnetApplicationTags.BACNET_APPLICATION_TAG_BOOLEAN, true)
```
or
```csharp
new BacnetValue(BacnetApplicationTags.BACNET_APPLICATION_TAG_ENUMERATED, 1)
```

chenzx1020, 2017-04-28:
Hi @gralin, i use the 
`new BacnetValue(BacnetApplicationTags.BACNET_APPLICATION_TAG_ENUMERATED, 1)` 
sent data for success,
however i use 
`new BacnetValue(BacnetApplicationTags.BACNET_APPLICATION_TAG_BOOLEAN, true)`
not success, 
thank you very much!

gralin, 2017-04-28:
@chenzx1020 I'm glad it works for you. If you find this library useful, a star on the repository will be apriciated :)

closed 2017-04-28 by gralin