ela-compil/BACnet #5: Hi the whois return 255.255.255.255

pigwing, 2017-08-05:
in my compuer whois return UnicastAddresses 255.255.255.255
and i send the correct who is length is 54 bug it is 50

pigwing, 2017-08-05:
in my computer Broadcast is not found , it is default 255.255.255.255, so i must bacnet_client.WhoIs(receiver: new BacnetAddress(BacnetAddressTypes.IP, "192.168.199.255", 0xFFFF));

pigwing, 2017-08-05:
GetBroadcastAddress has bug

pigwing, 2017-08-05:
but IPAddress.Broadcast is 255.255.255.255

gralin, 2017-08-08:
@pigwing thanks for the info. I'm swamped at work so can't look it up but if you could send me a pull request I will investigate.

gralin, 2017-08-09:
@pigwing I took a loot at the code that generates the broadcast address and it works for me. Are you setting your local endpoint address when creating the client object?

gralin, 2017-11-05:
@pigwing do you have any more info about the bug you are refering to or can I close this?

gralin, 2017-11-15:
Closed due to inactivity

closed 2017-11-15 by gralin