ela-compil/BACnet #7: Abort from device: 4

ReiminkB, 2017-09-30:
For a couple of devices a try to talk with, I get the following error when using the ReadPropertyRequest function:  "Abort from device: 4". I looked the byte up and I saw that it means: Invalid APDU in this state. 
Can I adjust this state in the driver?

gralin, 2017-11-05:
Hi @ReiminkB, can you post the part where you call ReadPropertyRequest and tell me what kind of object and property you want to read from?

gralin, 2017-11-15:
Closing due to lack of feedback

gralin, 2017-11-17:
@ReiminkB thanks for the feedback. Could you please try the latest beta (1.0.19) and see if you get a more human friendly message?[ Abort code 4 means that segmentations is not supported](https://github.com/ela-compil/BACnet/blob/dce2b87da5f4eca897b6b8a483fc665e20fbdfa1/Base/BacnetAbortReason.cs#L40). Your device has too many objects and you can't read them due to lack of segmentation support (read more about it [here](http://store.chipkin.com/articles/segementation-in-bacnet)). You need to read one object at a time using indexes.

Try this (I just typed it here so no guarantee it works but you will get the idea):

```csharp
var propertyId = BacnetPropertyIds.PROP_OBJECT_LIST;
client.ReadPropertyRequest(device.Ip, objectId, propertyId, out value_list, arrayIndex: 0);
var objectCount = value_list.First().As<int>();

for (uint i = 1; i <= objectCount; i++)
{
    client.ReadPropertyRequest(device.Ip, objectId, propertyId, out value_list, arrayIndex: i);
    object_list.AddLast(value_list.First().As<BacnetObjectId>());
}
```

ReiminkB, 2017-11-17:
@gralin thanks for the answer. I replace 1.0.13 for 1.0.19-beta and got indeed the message: "Abort from device, reason: SEGMENTATION_NOT_SUPPORTED"

Thanks for the useful information from that link.

I adjusted my code by reading the properties one by one, as you suggested, when trying to read the complete list fails.

I tested the code and now it works for that device as well.

Thanks!

gralin, 2017-11-17:
@ReiminkB I'm glad it works for you! If you find this lib useful please give it a star :innocent: I'm planning a rewrite of the API so it's more user friendly, so stick around for v2 in near future.

closed 2017-11-17 by gralin

TheKPAXian, 2019-03-28:
So, is there no way to read more than one object at a time, let's say, read in chunks of 10 objects?

petferns, 2021-04-26:
Hi @gralin I am trying to use the sample project (https://github.com/ela-compil/BACnet.Examples) BacnetToDatabase project. When I run, the form opens and I see ipaddress and name, I select it and click on Transfer to SQL CE, I get error - **{"Abort from device: 4"}** on this line of the code
bacnet_client.ReadPropertyRequest(device.Key, new BacnetObjectId(BacnetObjectTypes.OBJECT_DEVICE, device.Value), BacnetPropertyIds.PROP_OBJECT_LIST, out value_list);