ela-compil/BACnet #8: Weekday in trendlog

BennoMeijer, 2017-11-30:
Hi, 
When I use a trend log object and I try to read the data I'm getting weekday errors. In wireshark you can see that sunday is given day 0 instead of 7. 

I have changed something in the Serialize/ASN1.cs,  encode_bacnet_date

        public static void encode_bacnet_date(EncodeBuffer buffer, DateTime value)
        {
            if (value == new DateTime(1, 1, 1)) // this is the way decode do for 'Date any' = DateTime(0)
            {
                buffer.Add(0xFF);
                buffer.Add(0xFF);
                buffer.Add(0xFF);
                buffer.Add(0xFF);
                return;
            }

            /* allow 2 digit years */
            if (value.Year >= 1900)
                buffer.Add((byte)(value.Year - 1900));
            else if (value.Year < 0x100)
                buffer.Add((byte)value.Year);
            else
                throw new Exception("Date is rubbish");

            buffer.Add((byte)value.Month);
            buffer.Add((byte)value.Day);
            int DayOfWeek = (int)value.DayOfWeek;
                if (DayOfWeek == 0) DayOfWeek = 7;
            buffer.Add((byte)DayOfWeek);
        }
After I changed this I still cannot read a trendlog (only with YABE it works) , but my weekday errors are gone

gralin, 2017-11-30:
@BennoMeijer thank you for reporting this! I had a look at the specification and indeed you are right:

> fourth octet day of week (1..7)
> 1 = Monday
> 7 = Sunday
> 0xFF = unspecified

But what is interesting, YABE has the same bug:

https://sourceforge.net/p/yetanotherbacnetexplorer/code/HEAD/tree/trunk/Yabe/BACnetBase.cs#l4115

I will fix this as you suggested but it will not solve your problem right? I'm up for helping you but don't have any device I could query for trendlog... Any ideas?

BennoMeijer, 2017-12-01:
Hi gralin

When I look at the wireshark data I see something that is different from working devices that I have here.
The datatype doesn't look right to me,
[image]

When I change in System.IO.Bacnet.Serialize
ASN1.encode_tag(buffer, (byte)record.type, **false**, (uint)tmp1.offset); (line 2313) the bool bool contextSpecific in true, it looks the same as my other devices
[image]

gralin, 2017-12-01:
@BennoMeijer good catch! Since `BacnetTrendLogValueType` is being used tag number instead of `BacnetApplicationTags`, therefor we need to signal this as context specific (value `true`). In your case `TL_TYPE_REAL` (value 2) was treated as `BACNET_APPLICATION_TAG_UNSIGNED_INT` (same value 2) and that's what you were getting in first screenshot. This bug is inherited from YABE:

https://sourceforge.net/p/yetanotherbacnetexplorer/code/HEAD/tree/trunk/Yabe/BACnetBase.cs#l8383

BennoMeijer, 2017-12-05:
Hi Gralin
I have it almost working. After I send the historical data I need to send the next sequence number. Is there already a function in ASN1 I can use to create this last part?
[image]

gralin, 2017-12-06:
Hi @BennoMeijer, could you explain me your use case, what are you trying to do and where are you facing problem? Are you using `BacnetClient` class or just `ASN1` directly?

BennoMeijer, 2017-12-07:
Hi,

I write a program that acts as a bacnet device to import historical data as
counter values ​​by retrieving them by a rest api.
I started with code from the sample code of 'AnotherStorageImplementation'
and I use the NuGet package, latest version.
I have only overwritten the functions I mentioned above.

Regards

Benno Meijer

gralin, 2017-12-13:
@BennoMeijer is this solved or should this case be reopened?

BennoMeijer, 2017-12-13:
Hi,
if I add a value to a trend log, the status is uint. I think that should be a BacnetBitstring. When i try to set some value to the status i'm getting the wrong response.

[image]

gralin, 2017-12-14:
@BennoMeijer could you anwser a few questions so that I fully understand your issue?

1. You are using BACnet library + AnotherStorageImplementation project from examples to show a Trend object in your BACnet network that holds some values you get from other source through rest api, correct?

2. In above screenshot Trend object has invalid Status flags. This is the Trend object served by your project and your reading it with some other tool?

3. What did you mean by saying below? You mean by creating a new TrendLogRecord you were able to provide a big value which is later seen as invalud Status flags? If so, please check out the commit I referenced.

> When i try to set some value to the status

BennoMeijer, 2017-12-14:
Hi
In response to you question that's exactly what i try to do.
I setting the status from the trendlog to '3' to see if i get correct status flags but somewhere the uint is converted to a bitstring and something strange happens, I think it shout be a bitstring from the beginning. I'm setting the value trough the trendlog.AddValue().
I reading the trend by some kind of scada system for building management.
I will have a look at the commit and will come back to you

BennoMeijer, 2017-12-14:
Unfortunately the 'var recordStatusFlags = BacnetBitString.ConvertFromInt((uint)record.statusFlags);' does not work as i expected. When i set status flag 'IN_ALARM' it is still converted as zero

[image]

[image]

(I have only just copied the EncodeLog Record to my own code for debugging)

BennoMeijer, 2017-12-14:
Hi,
If i change the code into : 
            /* Tag 2: status */
            var recordStatusFlags = BacnetBitString.ConvertFromInt((uint)record.statusFlags);
            //if (recordStatusFlags.bits_used > 0)
            if (true)
            {
                recordStatusFlags.SetBit(0, false);
                recordStatusFlags.SetBit(1, true);
                recordStatusFlags.SetBit(2, false);
                recordStatusFlags.SetBit(3, true);
                ASN1.encode_opening_tag(buffer, 2);
                ASN1.encode_application_bitstring(buffer, recordStatusFlags);
                ASN1.encode_closing_tag(buffer, 2);
            }
the wireshark gives me the following 

[image]

BennoMeijer, 2017-12-14:
Hi,
It didn't work for me, but i found something. I have removed the opening and closing tags,

            /* Tag 2: status */
            var recordStatusFlags = BacnetBitString.ConvertFromInt((uint)record.statusFlags, 4);
            if (recordStatusFlags.bits_used => 0)
            {
                recordStatusFlags.SetBit(0, false);
                recordStatusFlags.SetBit(1, false);
                recordStatusFlags.SetBit(2, false);
                recordStatusFlags.SetBit(3, false);
                encode_application_bitstring(buffer, recordStatusFlags);
            }

and i had to change the call to encode_tag, set the tagNumber to BACNET_APPLICATION_TAG_UNSIGNED_INT end set the contextSpecific to true

        public static void encode_application_bitstring(EncodeBuffer buffer, BacnetBitString bitString)
        {
            uint bitStringEncodedLength = 1; /* 1 for the bits remaining octet */

            /* bit string may use more than 1 octet for the tag, so find out how many */
            bitStringEncodedLength += bitstring_bytesUsed(bitString);
            encode_tag(buffer, (byte)BacnetApplicationTags.BACNET_APPLICATION_TAG_UNSIGNED_INT, true, bitStringEncodedLength);
            encode_bitstring(buffer, bitString);
        }
now the wireshark looks ok,

[image]

closed 2017-12-19 by BennoMeijer