ela-compil/BACnet #9: Cant figure out how to trigger event notification

cemremengu, 2017-12-06:
I tried everything (including setting an event enrollment with a notification class) but cannot figure out how to trigger event notification

`bacnet_client.OnEventNotify += new BacnetClient.EventNotificationCallbackHandler(handler_OnEventNotify);`

Is this the right way to use it? Any help is greatly appreciated and thanks for this great library!

gralin, 2017-12-07:
Thanks @cemremengu. So did setting event enrollment work? If so you should start receiving events from your device. Best way to debug is to use [Wireshark with bacnet filter](https://wiki.wireshark.org/Protocols/bacnet). This way you will see if you are sending the enrollment, if it's being set and if you receive any events. You can also post your code here or PM me and I will take a look.

cemremengu, 2017-12-07:
Great, it was the simulator not producing event I think. It works on a real controller. Is there any way to get the `presentValue` property in the EventData?

gralin, 2017-12-07:
@cemremengu, I took a look at the code and decoding of some notification data is missing, take a look here:

https://github.com/ela-compil/BACnet/blob/master/Serialize/Services.cs#L870

Lower in that part of code you can see that encoding of the values is implemented so if you send, let's say notification of tyoe `BacnetEventTypes.EVENT_CHANGE_OF_VALUE`, the changed value will be encoded and sent to the recipient. So we are missing the implementation of the decoding. As a workaround I suggest that you read the present value whenever you get an event, or make a pull request to fix the missing part 😄

cemremengu, 2017-12-07:
Thanks for the suggestions! I will probably go for a PR :) The problem with sending a read request is that if the event is initiated by an `event enrollment` object, it does not tell me what the monitored object was so I have no information to query the value (unless I am missing something)

gralin, 2017-12-07:
PR is always welcome :) In event data you have two properties:

```csharp
public struct BacnetEventNotificationData
{
    public BacnetObjectId initiatingObjectIdentifier;
    public BacnetObjectId eventObjectIdentifier;
...
```

Is neither of these your monitored object id? If not, then you are right, you have to decode the missing part...

closed 2017-12-14 by cemremengu