ela-compil/BACnet #10: Segmentation NotSupported Exception

BennoMeijer, 2017-12-12:
Hi,

I'm getting a exception, probably caused by segmentation.
there is a timeline for the implementation of segmentation ?

System.Transactions Critical: 0 : <TraceRecord xmlns="http://schemas.microsoft.com/2004/10/E2ETraceEvent/TraceRecord" Severity="Critical"><TraceIdentifier>http://msdn.microsoft.com/TraceCodes/System/ActivityTracing/2004/07/Reliability/Exception/Unhandled</TraceIdentifier><Description>Unhandled exception</Description><AppDomain>BacnetConsoleApp.exe</AppDomain><Exception><ExceptionType>System.NotSupportedException, mscorlib, Version=4.0.0.0, Culture=neutral, PublicKeyToken=b77a5c561934e089</ExceptionType><Message>Specified method is not supported.</Message><StackTrace>   at System.IO.BACnet.BacnetTransportBase.set_MaxInfoFrames(Byte value)
   at System.IO.BACnet.BacnetClient.&amp;lt;&amp;gt;c__DisplayClass270_0.&amp;lt;HandleSegmentationResponse&amp;gt;b__0(Object o)
   at System.Threading.QueueUserWorkItemCallback.WaitCallback_Context(Object state)
   at System.Threading.ExecutionContext.RunInternal(ExecutionContext executionContext, ContextCallback callback, Object state, Boolean preserveSyncCtx)
   at System.Threading.ExecutionContext.Run(ExecutionContext executionContext, ContextCallback callback, Object state, Boolean preserveSyncCtx)
   at System.Threading.QueueUserWorkItemCallback.System.Threading.IThreadPoolWorkItem.ExecuteWorkItem()
   at System.Threading.ThreadPoolWorkQueue.Dispatch()
   at System.Threading._ThreadPoolWaitCallback.PerformWaitCallback()</StackTrace><ExceptionString>System.NotSupportedException: Specified method is not supported.
   at System.IO.BACnet.BacnetTransportBase.set_MaxInfoFrames(Byte value)
   at System.IO.BACnet.BacnetClient.&amp;lt;&amp;gt;c__DisplayClass270_0.&amp;lt;HandleSegmentationResponse&amp;gt;b__0(Object o)
   at System.Threading.QueueUserWorkItemCallback.WaitCallback_Context(Object state)
   at System.Threading.ExecutionContext.RunInternal(ExecutionContext executionContext, ContextCallback callback, Object state, Boolean preserveSyncCtx)
   at System.Threading.ExecutionContext.Run(ExecutionContext executionContext, ContextCallback callback, Object state, Boolean preserveSyncCtx)
   at System.Threading.QueueUserWorkItemCallback.System.Threading.IThreadPoolWorkItem.ExecuteWorkItem()
   at System.Threading.ThreadPoolWorkQueue.Dispatch()
   at System.Threading._ThreadPoolWaitCallback.PerformWaitCallback()</ExceptionString></Exception></TraceRecord>

gralin, 2017-12-12:
Hi @BennoMeijer, sorry for this, it was my fault. The original code was silently ignoring the value being set on MaxInfoFrames so I changed it to throwing an exception. This caused however that segmentation got broken (but it's implemented and supported). I've fixed it. Can you check if it works for you now?

BennoMeijer, 2017-12-13:
Yep, This fixes my exception.
thanks for your fast solution

closed 2017-12-13 by BennoMeijer

gralin, 2017-12-13:
@BennoMeijer glad it works for you. If you have any other problems or requests just open an issue and I will see what I can do. Also, if you find this lib useful, a star on the repo will be appreciated 😄 If you have some spare time, please check out the v2 branch, I'm trying to rewrite the API so it's more friendly there.

sevenam, 2020-03-12:
In my case I had not set the MaxSegments property for the BacnetClient. E.g.:

`BacnetClient.MaxSegments = BacnetMaxSegments.MAX_SEG4;`