ela-compil/BACnet #13: First stab at decoding notification values

cemremengu, 2017-12-18:
As mentioned in issue #9, here is my first stab in an attempt to decode the notification values, I used the bacnet-stack C library implementation as a direct conversion. The modification of `BacnetPropetyState` causes a breaking change for `bacapp_encode_property_state`. 

Any help, ideas, suggestions appreciated to take this forward.

----
**Services.cs**
```
/* tag 12 - event values */
switch (eventData.notifyType)
{
	case BacnetNotifyTypes.NOTIFY_ALARM:
	case BacnetNotifyTypes.NOTIFY_EVENT:

		if (ASN1.decode_is_opening_tag_number(buffer, offset + len, 12))
			len++;
		else
			return -1;

		if (ASN1.decode_is_opening_tag_number(buffer, offset + len, (byte)eventData.eventType))
			len++;
		else
			return -1;

		switch (eventData.eventType)
		{
			case BacnetEventTypes.EVENT_CHANGE_OF_STATE:

				len += ASN1.bacapp_decode_context_property_state(buffer, offset + len, 0, out var changeOfState_newState);
				break;
		}

		break;

	//In cases other than alarm and event
	//there's no data, so do not return an error
	//but continue normally
	case BacnetNotifyTypes.NOTIFY_ACK_NOTIFICATION:
	default:
		break;
}
```
----

**ASN1.cs**
```
public static int bacapp_decode_context_property_state(byte[] buffer, int offset, byte tagNumber, out BacnetPropetyState value)
{
	int len = 0;
	int sectionLength;
	value = new BacnetPropetyState();

	if (decode_is_opening_tag_number(buffer, offset + len, tagNumber))
	{
		len++;
		sectionLength = bacapp_decode_property_state(buffer, offset + len, out value);

		if (sectionLength == -1)
		{
			len = -1;
		}
		else
		{
			len += sectionLength;

			if (decode_is_closing_tag_number(buffer, offset + len, tagNumber))
			{
				len++;
			}
			else
			{
				len = -1;
			}
		}
	}
	else
	{
		len = -1;
	}

	return len;
}

private static int bacapp_decode_property_state(byte[] buffer, int offset, out BacnetPropetyState value)
{
	int len = 0;
	uint len_value_type;
	int section_length;
	uint enumValue;
	byte tagnum;

	value.tag = 0;
	value.state = new BacnetPropetyState.State();

	section_length = decode_tag_number_and_value(buffer, offset + len, out tagnum, out len_value_type);

	if (section_length == -1)
	{
		return -1;
	}

	value.tag = 0;

	value.tag = (BacnetPropetyState.BacnetPropertyStateTypes)tagnum;
	len += section_length;

	switch (value.tag)
	{
		case BacnetPropetyState.BacnetPropertyStateTypes.BOOLEAN_VALUE:
			value.state.boolean_value = len_value_type == 1 ? true : false;
			break;

		case BacnetPropetyState.BacnetPropertyStateTypes.BINARY_VALUE:
			section_length = decode_enumerated(buffer, offset + len, len_value_type, out enumValue);

			if (section_length == -1)
			{
				return -1;
			}

			value.state.binaryValue = (BacnetPropetyState.BACNET_BINARY_PV)enumValue;
			break;

		case BacnetPropetyState.BacnetPropertyStateTypes.EVENT_TYPE:

			section_length = decode_enumerated(buffer, offset + len, len_value_type, out enumValue);

			if (section_length == -1)
			{
				return -1;
			}
			value.state.eventType = (BacnetEventTypes)enumValue;
			break;

		case BacnetPropetyState.BacnetPropertyStateTypes.STATE:

			section_length = decode_enumerated(buffer, offset + len, len_value_type, out enumValue);

			if (section_length == -1)
			{
				return -1;
			}
			value.state.state = (BacnetEventStates)enumValue;
			break;

		case BacnetPropetyState.BacnetPropertyStateTypes.UNSIGNED_VALUE:

			section_length = decode_unsigned(buffer, offset + len, len_value_type, out value.state.unsignedValue);

			if (section_length == -1)
			{
				return -1;
			}
			break;

		default:
			return -1;
	}

	len += section_length;

	return len;
}
```
----

**BacnetPropertyState.cs**
```
public struct BacnetPropetyState
{
	public enum BacnetPropertyStateTypes
	{
		BOOLEAN_VALUE,
		BINARY_VALUE,
		EVENT_TYPE,
		POLARITY,
		PROGRAM_CHANGE,
		PROGRAM_STATE,
		REASON_FOR_HALT,
		RELIABILITY,
		STATE,
		SYSTEM_STATUS,
		UNITS,
		UNSIGNED_VALUE,
		LIFE_SAFETY_MODE,
		LIFE_SAFETY_STATE
	}

	public enum BACNET_BINARY_PV
	{
		MIN_BINARY_PV = 0,  /* for validating incoming values */
		BINARY_INACTIVE = 0,
		BINARY_ACTIVE = 1,
		MAX_BINARY_PV = 1,  /* for validating incoming values */
		BINARY_NULL = 255   /* our homemade way of storing this info */
	}

	public BacnetPropertyStateTypes tag;
	public State state; // !!!BREAKING CHANGE!!!! for encoding bacapp_encode_property_state

	public struct State
	{
		public bool boolean_value;
		public BACNET_BINARY_PV binaryValue;
		public BacnetEventTypes eventType;
		public BacnetEventStates state;
		public uint unsignedValue;
	}

	public override string ToString()
	{
		return $"{tag}:{state}";
	}
}
```

gralin, 2017-12-18:
@cemremengu thanks, I will take a look at it after work and will let you know. Could you open a pull request with your changes so it's easier for me to track changes? Later on it will be also easier to merge them and you will be included in contributors here.

cemremengu, 2017-12-18:
@gralin let's continue from  #14

closed 2017-12-18 by cemremengu