ela-compil/BACnet #24: More tests needed

DarkStarDS9, 2018-02-26:
Now that we started to add tests, we should add more. Currently, DotCover reports the following coverage:

[image]

It should be relatively easy to improve on this with back to back tests (like in *EventNotificationTests/should_raise_oneventnotify_when_sending_changeoflifesafety_data*) which provides much coverage with little effort.

To make sure that we are complying with the standard, more tests like *ServiceTests/should_encode_logrecord_according_to_ashrae_example* should be added - which can only be done by people having access to the documents.

Any takers? I'll work on additional tests for `EventNotifications`, covering other `EventValues` next. I'd suggest to comment here **who** is working on **what** to coordinate our efforts, but @gralin may have a better idea?! :wink:

gralin, 2018-02-27:
I support the initiative 👍 Lack of unit tests in the original code slowes down any changes but also some parts of the code just simply don't work and either no one uses it, or people fix them without openning a PR. If there were many active contributors we would fill up this hole in no time, but under current circumstances I guess it will take some time, I am up for it however! Aside of writing who is doing what we can create a TODO list here which namespaces/classes have 100% coverage and close the issue once we reach 100% everywhere.

DarkStarDS9, 2018-02-27:
I've covered `EventValues` and added tests for ASHRAE examples F.1.2 to F.1.8  in my working branch (except F.1.7 - GetEnrollmentSummary, because it doesn't seem to be implemented), the current stats are as follows:

[image]

I'll continue to add tests for more examples.

scottpidzarko, 2022-11-14:
Have these tests been added in a PR? This has been an open issue for almost 5 years and it seems odd that there has been no updates or closing of the issue.