ela-compil/BACnet #29: NetStandard Version

jferreirasetesys, 2018-11-14:
It would be a good solution to have a Net Standard version of the SDK. I think the only problem will be with serial port. One nice solution would be to have a base DLL compiled with Net Standard, and a Net 4.0 DLL based on that solution, adding the Serial Port MSTP implementation.
It could allow the Bacnet IP implementation on Linux, Mac, Xamarin, etc.

gralin, 2018-11-14:
@jferreirasetesys I agree, .NET Standard is the way to go with new projects, and libraries like this should be made compatible. Is anyone willing to contribute to this?

jferreirasetesys, 2018-11-14:
I actually already compiled it sucessfuly. It is just a matter of removing the MSTP components. I think the problem is more on the way it should be organized. With that solved i will gladly help.

maikebing, 2019-02-21:
#31  Update to netstandard2 and new project format

closed 2021-08-23 by gralin