ela-compil/BACnet #32: Receiving events (OnEventNotify) is not working

DoctorSloww, 2019-03-01:
With **BacnetClient** I have used **OnEventNotify** event, but it seems that it is never called. However, **OnUnconfirmedServiceRequest** does work.
I have used "AnotherStorageImplementation" example which is sending those events.

I think the problem lies in Serialize\Services.cs DecodeEventNotifyData method.

gralin, 2019-03-01:
hi @DoctorSloww, which version of the nuget are you using? Please try with latest beta. If that still exists, can you checkout the v2 branch and try again?

DoctorSloww, 2019-03-04:
@gralin, the nuget I am using is: NuGet Package Manager 4.6.0.
And I installed BACnet.Core, version 1.0.30.
I think I have no option for beta.

I tried with v2 branch and it is the same.
Is it possible that AnotherStorageImplementation example is not working properly?
How would you test this?