ela-compil/BACnet #34: How to use ReadPropertyMultipleRequest

mcmayer, 2019-03-18:
I'm starting to go round in circles with `ReadPropertyMultipleRequest`...

My understanding is that I need to pass in a `BacnetAddress` and a `List<BacnetReadAccessSpecification>`. Each `BacnetReadAccessSpecification` has a `BacnetObjectId` and a `List<BacnetPropertyReference>`. And a `BacnetPropertyReference` is a bacnet property ID (a `uint`, e.g. 85 for `PRESENT_VALUE`) and an array index.

So, as an example, if I wanted to get `PROP_PRESENT_VALUE`, `PROP_DESCRIPTION`, `PROP_UNITS` for some `ANALOG_VALUE` of instance `0` I *think* I should do this:

    var props = new List<BacnetReadAccessSpecification>();
    var adr = new BacnetAddress(BacnetAddressTypes.IP, "192.168.0.123:47808");
    IList<BacnetReadAccessResult> results;

    var objId = new BacnetObjectId(BacnetObjectTypes.OBJECT_ANALOG_VALUE, 0);

    var propRefs = new List<BacnetPropertyReference>();
    propRefs.Add(new BacnetPropertyReference(85, 0));    // PROP_PRESENT_VALUE
    propRefs.Add(new BacnetPropertyReference(28, 1));    // PROP_DESCRIPTION
    propRefs.Add(new BacnetPropertyReference(117, 2));   // PROP_UNITS

    var spec = new BacnetReadAccessSpecification(objId, propRefs);

    props.Add(spec);

    bacnet_client.ReadPropertyMultipleRequest(adr, props, out results);

This is always giving me an error `ERROR_CODE_PROPERTY_IS_NOT_AN_ARRAY`. I'm probably interpreting the purpose of the various objects the wrong way.

Can someone help out?

gralin, 2019-03-18:
@mcmayer How about doing it like this:

```csharp

var results = await bacnet_client.ReadPropertyMultipleAsync(adr, objId,
    BacnetPropertyIds.PROP_PRESENT_VALUE,
    BacnetPropertyIds.PROP_DESCRIPTION,
    BacnetPropertyIds.PROP_UNITS);
```

mcmayer, 2019-03-19:
I didn't know about these Async calls. I can give your suggestion a try.

But I still would like to know how to use these `BacnetPropertyReference` and `BacnetReadAccessSpecification` Lists.

jreisman, 2019-06-05:
I ran into the same problem when trying to use ReadPropertyMultipleRequest() and found that I had to change my BacnetPropertyReference from:
`new BacnetPropertyReference((uint) BacnetPropertyIds.PROP_PRESENT_VALUE, 0);`

To:
`new BacnetPropertyReference((uint) BacnetPropertyIds.PROP_PRESENT_VALUE, System.IO.BACnet.Serialize.ASN1.BACNET_ARRAY_ALL);`

I guess if you specify an index at all it wants to require it to be an array, but BACNET_ARRAY_ALL will let it accept non-array values.

gralin, 2019-06-06:
Correct, this is a design flaw made by the original authors. I totally agree that this is not intuitive, even though under the covers indeed the index is always sent. That's why I added some suplement methods (the async ones) to make life easier. In the future I was planning on breaking backward compatibility and rewritting the API so it makes more sense.