ela-compil/BACnet #36: Use Microsoft.Extensions.Logging instead of Common.Logging

basvanharten, 2019-04-08:
The Microsoft .Net extension Microsoft.Extensions.Logging seems to be the logical successor to the Common.Logging library. Since the extension is actively being developed in my opinion that is the better (future proof) solution

gralin, 2019-04-08:
@basvanharten thanks, I will look into this

LADSoft, 2024-06-21:
i have a serious issue with runaway threads which may or may not be in the bacnet stack....    i don't make threads in the project it is in so I'm thinking possibly it is something like an exception thrown in one of those threads created on the thread pool, but I really don't know.

So I would like to turn on logging but after several hours of research it came out that the logging currently in the bacnet stack is way out of date and won't do for .net core projects....

so it would be really helpful if this would be done...