ela-compil/BACnet #37: Bacnet BBMD Read request fail

gowtham31, 2019-05-01:
I am facing issue while connecting and reading Bacnet device objects through BBMD configuration. Though i get response of complex-Ack i am not able to read any object values. Please check the code below what i have used to read the bacnet object and wireshark packet details.
 public void ReadBacnetObject()
        {
            IList<BacnetValue> Value;
            BacnetIpUdpProtocolTransport bacnetIpUdpProtocolTransport = new BacnetIpUdpProtocolTransport(0xBAC0, true);
            bacnetIpUdpProtocolTransport.Start();
            bacnetIpUdpProtocolTransport.Bvlc.AddBBMDPeer(new IPEndPoint(IPAddress.Parse("10.155.60.3"), 47808), IPAddress.Parse("255.255.255.255"));
            bacnet_client = new BacnetClient(bacnetIpUdpProtocolTransport);
            bacnet_client.OnIam += new BacnetClient.IamHandler(handler_OnIam);
            //bacnet_client.RegisterAsForeignDevice("10.155.60.3", 30);

            //bacnet_client.RemoteWhoIs("10.155.60.3");
            //Thread.Sleep(20);
            if (bacnet_client.ReadPropertyRequest(new BacnetAddress(BacnetAddressTypes.IP, "10.155.60.23:47808"),  new BacnetObjectId(BacnetObjectTypes.OBJECT_DEVICE, 2000001), BacnetPropertyIds.PROP_OBJECT_LIST, out Value) == false)
            {
                foreach (var item in Value)
                {
                    Console.WriteLine($"Able to read data  {item.Tag} + {item.Value} ");
                }
            }
        }
[image]

Kindly request check if anything went wrong?

gralin, 2019-05-01:
@gowtham31 for sure this line is incorrect:

```csharp
if (bacnet_client.ReadPropertyRequest(..., out Value) == false)
```

just remove ``` == false```. Does this help? If not, does the answer in wireshark dump contain the object list?

gowtham31, 2019-05-03:
Thanks Gralin for your response.
Removing false check in ReadPropertyRequest doen't help. However i am getting the object list in the wireshark packet as per the below attachment
[image]
Also from debugging the library, i have noticed that events subscribed in BacnetAsyncResult from BeginReadPropertyRequest such as OnComplexAck are not triggered to complete the EndReadPropertyRequest .
 public bool ReadPropertyRequest(BacnetAddress adr, BacnetObjectId objectId, BacnetPropertyIds propertyId, out IList<BacnetValue> valueList, byte invokeId = 0, uint arrayIndex = ASN1.BACNET_ARRAY_ALL)
        {
            using (var result = (BacnetAsyncResult)BeginReadPropertyRequest(adr, objectId, propertyId, true, invokeId, arrayIndex))
            {
                for (var r = 0; r < _retries; r++)
                {
                    if (result.WaitForDone(Timeout))
                    {
                        EndReadPropertyRequest(result, out valueList, out var ex);
                        if (ex != null)
                            throw ex;
                        return true;
                    }
                    if (r < Retries - 1)
                        result.Resend();
                }
            }
            valueList = null;
            return false;
        }
Could you please check this part and let me know if i have missed anything?

MahajanHarsh, 2020-01-17:
I am facing the same issue. Did you find any solution to this?

gralin, 2020-01-23:
@MahajanHarsh can you share your code provide a wireshark dump of the relevant request and response?