ela-compil/BACnet #44: Raspberry Pi Example Has Exception - Platform not supported

JawadJaber, 2019-06-28:
I tried to run the RP example with Mono, and with .Net Core 2.2 Both trials return similar message at this line

  bacnet_client.Start(); 

platform not supported exception socket.iocontrol, handles Windows-specific control codes and is not supported on this platform.

hoangdungt2, 2019-08-01:
for i case it works just fine, when you create bacnet client do not enter ip address in there
`var bacnetClient = new BacnetClient(new BacnetIpUdpProtocolTransport(0xBAC0, false));`
`bacnetClient.start();`
I don't use mono, just publish as "linux-arm" and it will work well for rp3, e.g.
`dotnet publish -c Release -r linux-arm -o outputFolder`

JawadJaber, 2019-09-17:
The solution provided by @mykah89  and @pnguyen-ecs worked for me and not errors appeared.
Many thanks everybody.🙏

minzdrav, 2020-08-14:
Hi BACnet libary team
Please reopen issue and integrate fix for DisableConnReset to v2-netstandard branch.
I'm using .net core 3.1. Library working fine on windows 10 + wsl2 + docker and on raspberry pi 4 + ubuntu + docker.
IsOSPlatform is supported by .net core, .net framework, .net standard
https://docs.microsoft.com/en-US/dotnet/api/system.runtime.interopservices.runtimeinformation.isosplatform?view=netcore-3.1

`private static void DisableConnReset(UdpClient client)
        {
            bool isWindows = System.Runtime.InteropServices.RuntimeInformation.IsOSPlatform(OSPlatform.Windows);
            if (isWindows)
            {
                const uint IOC_IN = 0x80000000;
                const uint IOC_VENDOR = 0x18000000;
                const uint SIO_UDP_CONNRESET = IOC_IN | IOC_VENDOR | 12;

                client?.Client.IOControl(unchecked((int)SIO_UDP_CONNRESET),
                    new[] { System.Convert.ToByte(false) }, null);
            }
        }`

gralin, 2020-08-14:
Hi @minzdrav, you are right this should be integrated with that branch. I will let you know when this is available.

Sn3b, 2020-09-10:
Any idea when this change will be available?