ela-compil/BACnet #54: Can't JSON Deserialize BacnetAddress

drosenba, 2020-04-10:
I'd like to use  Newtonsoft.Json to serialize and then deserialize a List<BacnetAddress>
I can serialize it, but I can't deserialize it.   Why doesn't BacnetAddress have a **default** constructor?  Can it? :)

System.Reflection.TargetInvocationException: Exception has been thrown by the target of an invocation.
 ---> Newtonsoft.Json.JsonSerializationException: Unable to find a constructor to use for type System.IO.BACnet.BacnetAddress. A class should either have a default constructor, one constructor with arguments or a constructor marked with the JsonConstructor attribute. Path '[0].Address.net', line 1, position 19.

gralin, 2020-04-26:
@drosenba sure, this class could have all constructor arguments with default values which would solve your issue. Which branch are you using?

drosenba, 2020-04-27:
Nuget package BACnet.Core (1.0.20)

gralin, 2020-05-06:
@drosenba that package comes from a fork of this repository, becuase the original nuget from here still doesn't have .net core version up on nuget. I will try to publish it but since it was added in version 2 which has some breaking changes, it might be a beta at first.

https://www.nuget.org/packages/bacnet/

closed 2020-05-06 by gralin