ela-compil/BACnet #58: MSTP Read/write example

mwitt1337, 2020-08-30:
I didn't see any examples for read/write MSTP devices including the ip address, network number, MAC address, and instance id.

Can you create an example?

jgmdavies, 2020-09-09:
@mwitt1337  As no-one else has answered, have you seen the basic read/write example at [](https://github.com/ela-compil/BACnet.Examples) ?   I'm just trying to use it, but hitting some problems.
Jim

jgmdavies, 2020-09-09:
Thanks Jakub.  I hope I didn't mislead anyone about the MSTP - I'd noticed the commented-out call to 'BacnetMstpProtocolTransport' in that example!

I guess the code has been refactored since that line was written though, as it doesn't compile now, so I replaced it here with something like (simplified):

`
var serialTransport = new BacnetSerialPortTransport("COM4", 38400);
`
`
var transport = new BacnetMstpProtocolTransport(serialTransport);
`

Thank you for your example of MSTP with a 'TcpTransport', which I guess is what the OP actually wanted (rather than serial!).

Marwen09, 2021-02-22:
Hello I tried to create a bacnetAdress with MSTP protocol but there are bacnetAdress only for IP and Ethernet , any idea?

gralin, 2021-02-24:
As suggested in #70

closed 2021-02-24 by gralin