ela-compil/BACnet #59: useExclusivePort & dontFragment set to true don't work from Docker container

Sn3b, 2020-09-15:
I created a console app that runs from a docker container.
I have 2 bacnet gateways, one that doesn't require useExclusivePort & dontFragment, and it works fine, but the other one does require useExclusivePort & dontFragment to be set to true, and I keep getting the error "No response within 00:00:01".

I have no idea where to look, any suggestion appreciated!

gralin, 2020-09-16:
@Sn3b are you sure your container configuration allows UDP traffic on the ports used? Can you monitor with Wireshark what is being sent out and if indeed no response is being received?

Sn3b, 2020-09-16:
I have one of my gateways that doesn't require the use of useExclusivePort & dontFragment when creating the transport, and there it works fine with docker.

With the second one, the one that does require the use of useExclusivePort & dontFragment when creating the transport, and after a few tests, I see I do receive responses, but the source IP is the one of the internal docker bridge.

I haven't had time to test with the first gateway what the source IP is when I receive the responses, so I can't tell if this is the issue... I will let you know as soon as I have tested :)