ela-compil/BACnet #60: Priority Array read throwing ERROR CODE UNKNOWN PROPERTY

mwitt1337, 2020-10-19:
I'm able to read properties and values from the object list without issue until I try to read the priority array.

Code below and screenshot of console output attached.

    public BacnetDevice ReadDeviceProperties(BacnetDevice objDevice)
    {
      Bacnet_client.ReadPropertyRequest(objDevice.Address, new BacnetObjectId(BacnetObjectTypes.OBJECT_DEVICE, objDevice.InstanceId), BacnetPropertyIds.PROP_OBJECT_LIST, out IList<BacnetValue> value_list);
      LinkedList<BacnetObjectId> object_list = new LinkedList<BacnetObjectId>();
      StringBuilder strOut = new StringBuilder();

      foreach (BacnetValue value in value_list)
      {
        if (Enum.IsDefined(typeof(BacnetObjectTypes), ((BacnetObjectId)value.Value).Type))
          object_list.AddLast((BacnetObjectId)value.Value);
      }

      foreach (BacnetObjectId object_id in object_list)
      {
        //read all properties
        IList<BacnetValue> values = null;
        IList<BacnetValue> priority = null;
        IList<BacnetValue> names = null;
        bool listProps = true;
        

        try
        {
          if (!Bacnet_client.ReadPropertyRequest(objDevice.Address, object_id, BacnetPropertyIds.PROP_OBJECT_NAME, out names))
          {

          }

          if (!Bacnet_client.ReadPropertyRequest(objDevice.Address, object_id, BacnetPropertyIds.PROP_PRESENT_VALUE, out values))
          {
            
          }

          strOut.AppendLine("Device id:" + objDevice.InstanceId + " Property Name:" + names[0] + " Value:" + values[0]);

          if (!Bacnet_client.ReadPropertyRequest(objDevice.Address, object_id, BacnetPropertyIds.PROP_PRIORITY_ARRAY, out priority))
          {

          }

          if (listProps)
            foreach (BacnetValue v in priority)
            {
              strOut.AppendLine("Priority: " + v);
            }

          continue;
        }
        catch (Exception ex)
        {
          strOut.AppendLine("Exception:" + ex.Message);
          //perhaps the 'present value' is non existing - ignore
          continue;
        }
      }
      Console.Write(strOut.ToString());
      return objDevice;
    }

[image]

gralin, 2020-10-20:
Hi @mwitt1337, would you be able to provide me wireshark dump with the request and response (only those packets needed) which is triggering the error? This would be the easiest way to fix it. Thanks!

mwitt1337, 2020-10-20:
I'm trying to figure out what's going on here because there is a growing list of Property Ids that are returning Unknown Property.

I've attached a screenshot from BDT 2.12.01 Bacnet Explorer that returns values for these properties from the same devices that are returning Unknown Property using this API.

Of course the first instinct is to think that the device address is incorrect but shown in my screenshot in the initial post the property name and current value is returning without issue.

[image]

mwitt1337, 2020-10-20:
Here is my list of tested properties

**FAILED: UNKNOWN PROPERTY**
PROP_SYSTEM_STATUS
PROP_VENDOR_NAME
PROP_VENDOR_IDENTIFIER
PROP_MODEL_NAME
FIRMWARE_REVISION
PROP_APPLICATION_SOFTWARE_VERSION
PROP_PROTOCOL_SERVICES_SUPPORTED
PROP_PROTOCOL_OBJECT_TYPES_SUPPORTED
PROP_LOCAL_DATE
PROP_LOCAL_TIME
PROP_MAX_MASTER

**RETURNED VALUES**
PROP_OBJECT_NAME
PROP_PRESENT_VALUE
PROP_STATUS_FLAGS = 0000

mwitt1337, 2020-10-20:
> Hi @mwitt1337, would you be able to provide me wireshark dump with the request and response (only those packets needed) which is triggering the error? This would be the easiest way to fix it. Thanks!

Working on getting the Wireshark data

mwitt1337, 2020-10-20:
Dumped from code loop in this thread trying to read priority array.

[dump.zip](https://github.com/ela-compil/BACnet/files/5412110/dump.zip)

gralin, 2020-10-21:
Thanks @mwitt1337 I will look into this.

mwitt1337, 2020-10-23:
Update from more testing

I tested the property PROP_VENDOR_NAME and it returns a value for the device that responds to the WhoIs. The problem starts when I pull the PROP_OBJECT_LIST from the parent device who responds to the WhoIs and I try to read PROP_VENDOR_NAME from the devices in the PROP_OBJECT_LIST.

mwitt1337, 2020-11-16:
Can you at least verify the correct code to read the PROP_OBJECT_LIST from a device?