ela-compil/BACnet #62: Read Properties of devices on a BACNET IP router

mwitt1337, 2020-11-18:
Has anyone been able to successfully read properties from devices on a BACNET IP router?

I am able to read properties of the BACNET router and I can pull the PROP_OBJECT_LIST which returns all the instance ids of the devices on the router. However, when I try to read properties with the instance Ids of the devices from the PROP_OBJECT_LIST I get unknown property errors.

Am I doing something wrong? How do you read from devices attached to bacnet router?

digibum, 2021-06-29:
I to have issues with reading values from devices behind bacnet router. Any samples for this? Thank you!

mwitt1337, 2021-06-29:
The project was cancelled so I am no longer working on BACNET but given all the other issues with the project and the management it wouldn't surprise me if the devices and the BACNET router were not setup correctly.

closed 2021-06-29 by mwitt1337