ela-compil/BACnet #65: Could not run BACShark in parallel

dibrony, 2021-01-18:
Hi,

I am using this library.

When I use it as: BacnetIpUdpProtocolTransport udp_transport = new BacnetIpUdpProtocolTransport(c_bacnetPort,false) it works perfect but the problem is that I have 2 network cards on my PC and sometimes it sends the data through the wrong card.
I tried: BacnetIpUdpProtocolTransport udp_transport = new BacnetIpUdpProtocolTransport(c_bacnetPort,false,false,1472,bacnet_IP); here the data is sent on the specified card (mentioned in bacnet_IP) however it is reserving the network card (i am not able to launch the BacShark to check the result, I can see it via the yabe application, but I have to use BACShark...).

c_bacnetPort is always equal to 47808
bacnet_IP is the IP of my PC, let's say "10.110.45.20"
the other network card IP is: 2.1.1.1

Any help to get out of this please ?

Thank you
Rony

gralin, 2021-01-20:
Hi @dibrony, you are correct, by default the library will use any local interface to send the requests (at least for broadcasts but for unicast it will choose the right interface, so it shouldn't mix it). But if you specify the `localEndpointIp` for the transport, then you should limit it to that particular interface.

As to why it's blocking BACShark, it's because you are trying to use two apps and have them listen on the same port (47808). This is possible using Yabe and custom app using this library, becuase the original author has created a special mode where there is a shared connection for multiple applications to use, so you can also run multiple instances of Yabe when you don't change the Yabe settings to use exclusive port. If you just want to check if the communication is running fine, just use Wireshark and filter the traffic to bacnet.

BTW, what is BACShark? I couldn't find it with quick Google search. Is that some BACnet client I could download and test?

dibrony, 2021-01-20:
Hi,

Thank you @gralin for the above info, I have also checked your answer on my other issue (#66). I will try them within few hours.
However, I noticed something yesterday while trying my code: the data I sent through BACnet is reained locally, the other guy is not able to see them on his PC via the server (as he can see other machines parameters...). Do you think setting the useExclusivePort to true will solve this problem ?

As for the BACShark, I can share it with you for testing, it is around 4MB, how can I send it to you ?

Regards

dibrony, 2021-01-20:
Hi again,

Also there is something called network number (if i am not mistaken), we need it to be 100. How can I set it please ?
Just found your email, I will send you the program now.

Regards

gralin, 2021-01-20:
>Also there is something called network number (if i am not mistaken), we need it to be 100. How can I set it please ?

The API of this library is a bit tricky, that's something I didn't change and it's inherited from the original implementation inside YABE. So you declare the network value in the `BacnetAddress` which you later provide as agument to read/write values from remote objects.

```csharp
var address = new BacnetAddress(BacnetAddressTypes.IP, "192.168.1.100", network: 100);
```

> the other guy is not able to see them on his PC via the server (as he can see other machines parameters...). Do you think setting the useExclusivePort to true will solve this problem ?

Yes, could be, the other device might ignore packets coming from different port than 0xBAC0. Best way to verify is to run Wireshark on the server, filter by bacnet and see if the UDP packets are being received. There might also be some firewall issues along the way.

> As for the BACShark, I can share it with you for testing

Thanks, I will take a look at it later.

dibrony, 2021-01-20:
Hi,
Thanks again for your support, however please advise where should I put the code of the network number.

dibrony, 2021-01-25:
Hi again,
concerning the above, i remove the m_ip_server.OnSubscribeCOV += new BacnetClient.SubscribeCOVRequestHandler(OnSubscribeCOV); code. It seems to be working now, althought I don't know if I need it elsewhere...

On more thing, how can I create a parameter of type "Enumerated" ?

Thank you

dibrony, 2021-01-28:
Hi,

Thank you for your support, it seems to be working now.

However it is not sending the Alarm notification to the monitor, i think it is related to the receipient list property (Bacnet type: Destination). I am not able to add this property.

Can you help in this ? any example please ?

Regards

gralin, 2021-02-11:
Hi @dibrony can you please clarify, do you want to receive alarm notification in your code or do you want to send it?

dibrony, 2021-02-11:
Hi,

I want to send an alarm notification to the server (according to an input I have).
server instance=200

Thank you
Rony

gralin, 2021-02-11:
Ok and you are interested in

1. `client.Notify()` - sends change of value notification
2. `client.SendUnconfirmedEventNotification()` - sends event notification

closed 2021-04-23 by gralin