ela-compil/BACnet #66: How to change the Source port ?

dibrony, 2021-01-19:
Hi,

Each time I run the application i have a new random source port while the destination port is always 47808.
How can I fix the source port ? I need it to be 47808 as well.

Thank you

gralin, 2021-01-20:
Hi @dibrony, you need to set `useExclusivePort` on the transport to `true` and specify the port you want to use.

```csharp
var localPort = 0xBAC0;
var localEndpointIp = "192.168.1.2";

var transport = new BacnetIpUdpProtocolTransport(
    localPort, useExclusivePort: true, localEndpointIp: localEndpointIp);

var client = new BacnetClient(transport);
```

This is the same setting you need to change in Yabe the first time you use it, so it always uses the same source UDP port.

[image]

ameyaalpha, 2021-04-23:
Which BACnet protocol version does the stack support? Has any of the device using the stack gone through BTL testing? Can you please let us know?

gralin, 2021-04-23:
@ameyaalpha please open a new issue as this question is not related to this topic

closed 2021-04-23 by gralin

ameyaalpha, 2021-04-23:
OK thanks @gralin