ela-compil/BACnet #68: Method overload of ReadRangeRequest() for RR_BY_TIME missing

patrickthoma, 2021-02-17:
Hello,

I have the requirement to read trend log data starting from a certain point in time onwards. I can't use the `ReadRangeRequest()` method as I don't know from which index on I have to read data.

Therefore, I would like to ask if you could provide a method overload of the `ReadRangeRequest()` method, take a `DateTime from` parameter? Seems like `EncodeReadRange` already supports RR_BY_TIME, so it would basically be necessary to propagate this enum to the method signature.

Best regards,

patrickthoma, 2021-02-22:
I tried to solve the first issue by myself, see #69

gralin, 2021-02-24:
Hi @patrickthoma thank you for solving the issue you have reported.

As to the manual, unfortunately there is none. The story of this library is rather complicated, it was developed by different developers in different points in time, and last one (Frédéric Chaxel) created YABE app with it. When I first came across YABE it helped me investigate some BACnet device and later I wanted to write a small microservice that would handle the communication. Unfortunately, this codebase was not self-contained as library. What I have done was extraction of the part that should be a reusable library, refactored the code (originally it's stored mostly in two files, one of which is 15k lines), pushed it on GitHub (originally it's SVN repo on SourceForge) and pushed it via CI to NuGet for easy usage. After all this I have contacted YABE author about maybe YABE using this NuGet package, so that we can work on this part in a separate repo. In response I have received acusations of being a theft, of my refactor work being stupid, that my additions are worthless (I have added support for LifeSafety objects and some async methods to client) and I should delete everything.

Anyway, the API of the library is not great (imho) and there was no original documentation as the author focuses on developing the app and not distributing this as reusable library. The examples also come from his repo. I could generate some automatic documentation of the API if that would help.

gralin, 2021-02-25:
Your changes are available in v1.1.2 on NuGet, thank you!

closed 2021-02-25 by gralin