ela-compil/BACnet #75: Need Routing Info

LADSoft, 2021-03-11:
on BIP the standard way to handle multiple device objects internal to one physical server device seems to be to have a router and set the source/destination fields appropriately.   In particular packets sourced from the device holds a SOURCE field in the NPDU header and coming into the device the DESTINATION field in the NPDU header tells which device to route to (the router device itself reacts to an empty destination and fills in an empty source).

To send messages properly then I need access to the destination field on incoming messages; but I don't seem to have that.   F Chaxel added the incoming SOURCE address to the BacnetAddress structure (as the RoutedSource) and I'm proposing you do similar for the incoming DESTINATION address.   E.g add a RoutedDestination field in the BacnetAddress structure, and initialize it at the same time the RoutedSource field is initialized.   Then I can use the destination address to do the appropriate routing.

Technically speaking there may be further issues with actually sending the source on various messages (e.g. I'm not sure if I can specify a SOURCE for functions like SimpleAck or ComplexAck).    But I'm not so worried about that as I can do simple workarounds for those kinds of issues at least in the short term.

It would be really helpful though if you could add something like a RoutedDestination field to the BacnetAddress structure,   then I think I would have enough to be able to make a switch to the nuget package.

gralin, 2021-03-11:
Hi @LADSoft thank you for sharing. May I ask if you have already implemented this in your own fork? If yes, maybe you would be willing to open a pull request to have it merged here? I sounds like a valid use case. I'm just worried if I will be able to test it correctly and it seems that you already have a setup for this. If you didn't implement this yet or you are using a different library that covers this use case, then would you mind testing the PR before it's merged?

LADSoft, 2021-03-12:
Thanks, I checked and got the ok from my supervisor to do a PR for this project.   We are porting both a client and server from the C language BACnet stack by Karg to C# so I can test the server-side changes against the old version of our client... both had previously been to plugfest a couple of times so we are pretty confident in how they work.   Give me a couple of weeks to finish up the testing then I will push something.     I may try to add something else we need. if that is ok...   response processing for GetEventInformation...

LADSoft, 2021-03-19:
thank you @gralin for your effort in this, it does help not to have to host our own version.    I noticed you optimized a little too :smile:.

I ended up adding everything on our wishlist for the server side now...   that includes the change I talked about earlier as well as support for sending confirmed notifications, and server side support for event acknowledge and get event information.    oh and I added a simple dispatcher  for network messages that we need to support 'WHO_IS_ROUTER_TO_NETWORK/I_AM_ROUTER_TO_NETWORK'...    all it does is route network layer messages into C# events the same way the confirmed and unconfirmed messages are done.

I just have to test and make sure nothing is broken.    Probably within the next couple of weeks as I have to work on other projects for a while.

I was wondering, I don't think my version of the lib will be binary compatible with the old one...   the binary version of the api is likely to be modified by my changes because I did things like add a missing field to a structure, and add defaulted arguments to lots of functions.   But I do intend it to be backward compatible at the source code level.   Is that ok?

LADSoft, 2021-03-24:
added to the list:   a couple of minor bug fixes probably noone ran into before...   also sending of 'abort' and 'reject' error messages is being overhauled to work as per the spec.    Meanwhile I really want to change one of the reject reasons from 'RECOGNIZED_SERVICE' to 'UNRECOGNIZED SERVICE' but that would mean source-level incompatibility.    Sigh.

closed 2021-04-01 by LADSoft