ela-compil/BACnet #76: Bacnet Client not discovering on Diffrent networks

9008926625, 2021-03-18:
Hi 

I am using Bacnet client dll ver 1.1.4.0. Bacnet devices not discovering in yabe and BDT (Bacnet Discovery tool) on different Networks.
It is working for the same networks.

e.g 
1. I have 172.16.1.64 (Yabe Explorer) and 172.16.1.23 (Bacnet device)  it is working fine
2. 172.16.1.23 (Bacnet device) and 192.168.2.200 (Yabe Explorer)  not discovering Bacnet Devices 

Please do the needful.

gralin, 2021-03-18:
This should help: https://github.com/ela-compil/BACnet/issues/65#issuecomment-763702546

9008926625, 2021-03-18:
Thanks for your response.

where i need to add this in bacnet Library ?
receiver = new BacnetAddress(BacnetAddressTypes.IP, "192.168.2.200", network: 10000);

9008926625, 2021-03-18:
Now I'm able to see BACnet device name in yabe after adding this Code in I Am Function of BacNet Library

public void Iam(uint deviceId, BacnetSegmentations segmentation = BacnetSegmentations.SEGMENTATION_BOTH, BacnetAddress receiver = null)
        {
           

            receiver = new BacnetAddress(BacnetAddressTypes.IP, "192.168.2.200", network: 10000);
            var b = GetEncodeBuffer(Transport.HeaderLength);
            NPDU.Encode(b, BacnetNpduControls.PriorityNormalMessage, receiver);
            APDU.EncodeUnconfirmedServiceRequest(b, BacnetPduTypes.PDU_TYPE_UNCONFIRMED_SERVICE_REQUEST, BacnetUnconfirmedServices.SERVICE_UNCONFIRMED_I_AM);
           
            Services.EncodeIamBroadcast(b, deviceId, (uint)GetMaxApdu(), segmentation, VendorId);

            Transport.Send(b.buffer, Transport.HeaderLength, b.offset - Transport.HeaderLength, receiver, false, 0);
           
        }

Now I'm unable to retrieve the objects of in yabe. Screenshot attached for reference.

Do I have to add in handler_OnReadPropertyRequest Event also?
Please suggest

closed 2021-03-22 by 9008926625