ela-compil/BACnet #85: Any plan for supporting dotnet core/dotnet 5?

ztytotoro, 2021-06-10:
Thanks for your jobs!!!! This project is awesome and I am trying to use it in my dotnet core 3.1 app.

but i got an incompatible message, I noticed that this project is based on dotnet framework 4.

do you have any plan to provide a dotnet standard version?

Microsoft has provided a convenient tool can easily transfrom dotnet  framework to dotnet core

maybe [this](https://github.com/dotnet/upgrade-assistant) will help you.

msschl, 2021-06-11:
I would also be interested in this library targeting .net 5

msschl, 2021-06-11:
There is already a branch adding [netstandard2.0](https://github.com/ela-compil/BACnet/tree/v2-netstandard)

ztytotoro, 2021-06-11:
> There is already a branch adding [netstandard2.0](https://github.com/ela-compil/BACnet/tree/v2-netstandard)

Great! but seems that it has been stopped updating for months

msschl, 2021-06-17:
I've gone ahead and updated the csproj to the new sdk style and also added .net standard 2.0 as a new TFM next to net40.

See #86

gralin, 2021-06-17:
Hi @msschl thank you for your contribution! Sorry that there is not much going on in this project lately. I will test and merge or comment your PR over this weekend ok?

msschl, 2021-06-17:
Thanks 👍🏽 That sounds good

gralin, 2021-06-20:
This was long overdue but thanks to @msschl now the project targets both .NET Framework and .NET Standard so can be used with latest versions of .NET 😃 I've bumped the version to v2 as this is a major change. It stands in conflict with a branch named v2, but that branch will not be merged any time soon (main contributor changed a lot there but didn't finish, so we might only cherry pick from it at some point). We can rename it to something else so there is no confusion.

Anyone willing to test, the new version is avaialble as prerelease on NuGet: https://www.nuget.org/packages/BACnet/2.0.0-beta

closed 2021-06-20 by gralin

msschl, 2021-06-21:
Thanks @gralin 🚀