ela-compil/BACnet #87: Is there a more dumbed down version of the examples?

vseven, 2021-06-18:
I'm usually pretty decent at adapting add ins but I'm having trouble with this one.  I have a data logger program that I'm currently using with a Obix library and I'm successfully polling data from Tridium Jaces and web supervisors.  I'd like to poll some data from  BACNet controllers now.  I've successfully gotten a RemoteWhoIs to poll the BBMD and give me a device list but now I want to poll a device and get back all it's properties.  I can directly pull a property but is there a example of pulling all BACNet properties for a device as a list so I can see what properties exist?  I didn't see that example anywhere.

Padanian, 2021-06-18:
In vb.net I did it like this:

```
    Private Function GetPropertyArray(adr As BacnetAddress) As BacnetPropertyIds()
        Dim multi_value_list As IList(Of BacnetReadAccessResult) = {}
        Dim properties As BacnetPropertyReference() = New BacnetPropertyReference() {New BacnetPropertyReference(CUInt(BacnetPropertyIds.PROP_ALL), System.IO.BACnet.Serialize.ASN1.BACNET_ARRAY_ALL)}
        multi_value_list = ReadPropertyMultipleRequest(adr, deviceObjId, properties)

        If multi_value_list Is Nothing Then
            multi_value_list = ReadPropertyRequest(adr, deviceObjId, BacnetPropertyIds.PROP_ALL)
        End If

        Dim result() As BacnetPropertyIds = {}
        If multi_value_list IsNot Nothing Then
            For Each itm In multi_value_list(0).values
                Array.Resize(result, result.Length + 1)
                result(result.Length - 1) = itm.property.propertyIdentifier
            Next
        End If
        Return result
    End Function
```

You can easily convert it back to c#
Hope this helps

vseven, 2021-06-18:
That's great, my program is VB already.  Thank-you.

closed 2021-06-18 by vseven

Padanian, 2021-06-18:
Let me know whether it works for you. YMMV.

vseven, 2021-06-18:
It didn't but I'm gonna keep plugging away at it.