ela-compil/BACnet #93: Problems running on Linux

cuican6, 2021-12-07:
hi . sometimes, networkcard in linux has two or more ipaddr,  but in bacnet  =>BacnetIpUdpProtocolTransport.cs=>GetAddressDefaultInterface method，two or more ipaddress is  bad, in my submission，there  canbe  write              return unicastAddresses.Length >= 1
                ? unicastAddresses.First()
                : null;
 sir,tell me  my proposal is     correct？
other,i  binding ip in bacnetclient, BacnetClient client = new BacnetClient(.....,"192.168.1.200");in this ,i  send  whois ,can't receive  response , no event of oniam trigger，  but i confirm the package of whois   has  send, the other client (don't bind ip) has received message of i am.
think you.

pocketbroadcast, 2022-02-23:
Yes, I can confirm, this is a problem.

If there are multiple ip addresses configured for a single device the [condition](https://github.com/ela-compil/BACnet/blob/a6869708758e70dd196e618459bedf3844ea83dd/Transport/BacnetIpUdpProtocolTransport.cs#L374) evaluates to false and the result is null as if there was no interface at all.

In case of multiple ip addresses configured, which one should be chosen?
Any comments?

Nevertheless, resolving the default ip address is only used when we bind to any interface -> 0.0.0.0
-> imho if there are multiple ip addresses / subnets available this option is dangerous by default.

gralin, 2022-03-01:
Thank you @cuican6 for reporting and @pocketbroadcast for checking this. It seems this address is only used to guess the correct broadcast address in case there is only one available interface address, otherwise it should return null so that default broadcast address is used. The[ original YABE code](https://sourceforge.net/p/yetanotherbacnetexplorer/code/HEAD/tree/trunk/Yabe/BACnetTransport.cs#l367) still has the same logic.

@pocketbroadcast I've added a suggestion for your PR