ela-compil/BACnet #94: Contributions and Project Activity

shaggygi, 2021-12-11:
@gralin This project seems interesting and are some areas I would like to add PRs for.  However, it appears to not be as active of late along with not reviewing/accepting other PRs.

I wouldn't mind helping but if there are no plans to continue, I would rather fork to a different path.  Any thoughts are appreciated.

gralin, 2021-12-11:
Hi @shaggygi, thanks for your interest in helping out with this project. For the past two years I was overwhelmed by work related projects and didn't find enough time and motivation to contribute much or properly test PRs and merge them. Since the origin of this repository there has been a few people that showed some interest in helping with development but then stopped. If were to start development and show commitment I wouldn't mind adding you to maintainers so that you don't have to fork it. In one week I will take two weeks off from work (first time in two years) so I might try to come back to the OSS world and maybe help out here too.

shaggygi, 2021-12-11:
@gralin Understood as it can be difficult to keep up with a protocol like BACnet.  I created a similar .NET implementation (not OOS) a few years back and know the challenges.

As for PRs, I would like to first attempt a few simple updates related to new C# 10 features.  This would allow me to become familiar with the code structure and such.  I have a few BACnet devices so this would be a good opportunity to try communication with them, as well.

Thanks.

gralin, 2021-12-11:
@shaggygi BACnet is... a 'special' protocol 😉 Few questions:

1. Do you maybe have access to BACnet specification? That's something that a maintainer should have, to be able to add or refactor things. I have access to mine (ANSI/ASHRAE Standard 135-2016) through the company I work for
2. Great that you have access to BACnet devices. Would you be willing to also do some tests on Linux? Some people reported issues regarding that and maybe you could address them.

shaggygi, 2021-12-11:
@gralin 

> Do you maybe have access to BACnet specification? That's something that a maintainer should have, to be able to add or refactor things. I have access to mine (ANSI/ASHRAE Standard 135-2016) through the company I work for

Yes, I also have access to the spec.

> Great that you have access to BACnet devices. Would you be willing to also do some tests on Linux? Some people reported issues regarding that and maybe you could address them.

I can eventually try, but do more on Windows.  Lemme become more familiar with the code and communication first.  Then I'll see what I can do in this space.  Baby steps 😄

shaggygi, 2021-12-11:
@gralin As mentioned, #95 is a simple PR to get me started and help update a few new language features.  I'm going to turn focus on setting up device and communication for now.  Thx

closed 2021-12-14 by gralin