ela-compil/BACnet #121: SEGMENTATION NOT SUPPORTED WHEN REQUESTING DEVICE WITH MANY OBJECT PROPERTIES

progirisdev, 2023-02-07:
Purpose 
============
Requesting a device bacnet objects, when the device contains a lot of objects (INTESYS Bacnet Gateway). Each I/O connected on the several slaves on the LON part, is an object on the BACNET side.

Method
============
Request is done using the following command...
`var objectList = await Client.ReadPropertyAsync(_adr, BacnetObjectTypes.OBJECT_DEVICE, deviceid, BacnetPropertyIds.PROP_OBJECT_LIST);`

Problem
============
When requesting a "standard" device with few objects, no problem,
When requesting the INTESYS which can have dozens of objects, it throws an error with the SEGMENTATION_NOT_SUPPORTED reason.

Possible issue
===========
Found on Internet is the following possible issue ;

".....Segmentation is used for APDUs (responses) that are too large to fit in one frame on the datalink. If both devices support segmentation, then great. If one of the device does not support segmentation, and the APDU does not fit, then "Segmentation not supported" error is issued. (and you have to then retry a smaller request)...."

Question
==========
How do we play with APDU size matching between the device and the client so that it correctly read all the objects ?

YarekTyshchenko, 2023-02-09:
What I ended up doing is reading each property individually, as I've seen some devices say they support segmentation, but actually not doing it. Protocol has a feature if you request index 0 the result is count of items, which you can then loop over and get one by one

YarekTyshchenko, 2023-02-12:
You know it's ID, its in your example above, `PROP_OBJECT_LIST`. Some overloads of `ReadPropertyAsync` allow passing `arrayIndex`

progirisdev, 2023-02-13:
Hi Yarek,

Deviceid is for the device. If I want to play with the Array, I need to retrieve in any manner the Array itself, but without the defniintion of each object.... PROP_BOJECT_LIST retrieve all the stuff and that is my problem.

YarekTyshchenko, 2023-02-22:
I suggest you build a small emulator to try to replicate the problem, and play around with it. BACnet protocol has a few layers on top of it, some have extra meaning that isn't defined by the protocol itself. My previous comment still stands. Making a request with `arrayIndex` of `0` will return single result containing an integer, representing a count of items. I can't make it clearer than that.

chenjing1294, 2024-10-25:
@progirisdev 
```csharp
IList<BacnetValue> objectList = new List<BacnetValue>();
try
{
    objectList = await sender.ReadPropertyAsync(adr, deviceObjId, BacnetPropertyIds.PROP_OBJECT_LIST);
}
catch (Exception exception)
{
    if (!string.IsNullOrWhiteSpace(exception.Message) && exception.Message.Contains(nameof(BacnetAbortReason.SEGMENTATION_NOT_SUPPORTED)))
    {
        IList<BacnetValue> list = await sender.ReadPropertyAsync(adr, deviceObjId, BacnetPropertyIds.PROP_OBJECT_LIST, arrayIndex: 0);
        if (list.Count == 1)
        {
            if (list[0].Value is uint size && size > 0)
            {
                for (uint i = 0; i < size; i++)
                {
                    IList<BacnetValue> temp = await sender.ReadPropertyAsync(adr, deviceObjId, BacnetPropertyIds.PROP_OBJECT_LIST, arrayIndex: i);
                    foreach (BacnetValue value in temp)
                    {
                        objectList.Add(value);
                    }
                }
            }
        }
    }
    else
    {
        throw;
    }
}
```